queens-satellite-team/pcbs #61: BAT-DB: Hardware Revisions

SeanTedesco, 2023-05-25:
# Overview
The team has received the BAT-DB (battery board used with the EPS system to provide power storage for the satellite) and we are required to test it. This ticket is used to track the validation, issues, and improvements to the design of the BAT-DB. Comments to this ticket are given as the board undergoes its test and improvements are identified.

[image]

# Supporting Documents
- [Battery Board Requirements](https://queensuca.sharepoint.com/:w:/t/GROUP-QSET/EQtAdzk_EARIrMd4tkivvn8BwNerQp--YzVH_r26aHvoOQ?e=Hxiomy)
- [Battery Board Design](https://queensuca.sharepoint.com/:w:/t/GROUP-QSET/EWqJf31M83NMuHhEOlDTnKcBC-Sz5hyvzvNoyW-eYluKog?e=LJZ6em)
- [Battery Board Test Plan](https://queensuca.sharepoint.com/:w:/t/GROUP-QSET/EWVV9Ow1So5EnTouTD9SN1QBBHr0qF3FQgiTU-P9vqx81w?e=55cOZF)

griffinjones1, 2023-05-25:
c10 (0204 size) was lost in trying to repair a reflow error. decoupling capacitor for timer LTC6995 (U4). will function fine without but add back in future iterations.

griffinjones1, 2023-05-28:
power control schematic - change flight model and eng model jumper design to be the same for J7 and J9 in order to reduce confusion. Currently one is in parallel and the other is in series.

griffinjones1, 2023-05-28:
Many more test points needed - live list:
- BCR_OUT
- 5V CHG input
- PCM_IN
- GND

a clean list of TPs on the front page of schematic will be good for easy reference

griffinjones1, 2023-05-29:
watchdog and deployment timer pins are messed up and shorted to ground. fix these pins for next iteration

griffinjones1, 2023-05-29:
battery clips are too close together - clips contact each other and short batteries when installed.
also test connector pcb cannot fit in J3 when a battery is installed.