queens-satellite-team/pcbs #62: UMB-DB: Hardware Revisions

SeanTedesco, 2023-05-27:
The team has received the UMB-DB (a debugging toll for QSET hardware) and we are required to test it. This ticket is used to track the validation and improvements to the design of the UMB-DB. Comments to this ticket are given as the board undergoes its test and improvements are identified.

[image]

SeanTedesco, 2023-05-27:
During the sprint for competition the ADCS-CTO board was being codded using the UMB-DB and when set up later after bein taken apart it bricked. It was determined that the 3.3V had shorted. After discussion and review it was determined that the short was due to the UMB-DB was powered and was briefly connected wrong due the pins being finicky to connect and could slip into the wrong slots. The redesign should make the pins unable to be connected unless they are correctly positioned.

jillianmalherbe, 2023-05-30:
Next iteration of the debug board should feature an LED that shows when the board is receiving power.